kakopappa/sinric #406: Give Alexa response

Derrin616, 2020-04-02:
Hi,

Is it possible to let Alexa say someting after you triggger it? Like the temperature?

Nice project i use it all the day

Derrin616

Derrin616, 2020-04-14:
Can you please add this feature?

kakopappa, 2020-04-15:
hi, 

Sinric can not let Alexa say something. 

You can create a temperature sensor device in Sinric Pro (https://sinric.pro) and use the temperature sensor sketch to send the temperature. Sinric Pro has graphs support as well.

https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/temperaturesensor

When you ask Alexa, what's the temperature of .... then Alexa will respond

https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-temperaturesensor.html