kakopappa/sinric #408: Sinric pro devices not appearing in my alexa app

hemsinghBanoth, 2020-04-09:
Hello,
I have paid for 10 devices , but 
1.  sinric devices only appearing in Alexa app. 
2. I have removed some of the devices from Sinric but still working 
3. Sinric pro devices not appearing in Alexa App. 

 pl. solve these issues. As i want to pay further for 20 devices .  I want to have a permanent solution  for theses problems such that i can further work upon. 

Regards,
 Hem.  cmsscr@yahoo.com.

hemsinghBanoth, 2020-04-09:
> Hello,
> I have paid for 10 devices , but
> 
> 1. sinric devices only appearing in Alexa app.
> 2. I have removed some of the devices from Sinric but still working
> 3. Sinric pro devices not appearing in Alexa App.
> 
> pl. solve these issues. As i want to pay further for 20 devices . I want to have a permanent solution for theses problems such that i can further work upon.
> 
> Regards,
> Hem. [cmsscr@yahoo.com](mailto:cmsscr@yahoo.com).
Just now I have added sinric pro skills to alexa. Now I found Sinric pro devices in alexa.  But  how to use both sinric and sinric pro devices in a single arduino esp program ?

kakopappa, 2020-04-09:
I never tried running both. Try putting both in the same sketch. I am not sure whether Websocket library can handle multiple connections at the same time to two different servers