kakopappa/sinric #414: Read Temperature from Dht11 and send it to Alexa

Haxhi03, 2020-04-22:
Is it possible to read temperature and send it to Alexa?
I looked at some old issues but i didn't find the answer.
I use the code of Thermostat with Dht11.ino

sivar2311, 2020-04-23:
You have to take a closer look at the thermostat with dht11 example.
There is a [setSetTemperatureSettingOnServer function](https://github.com/kakopappa/sinric/blob/afb283457a142a124d0fcbc928782e502967e858/arduino_examples/thermostat_example_with_dht_11.ino#L229) which sends the current ambient temperature and humidity to the server.

Haxhi03, 2020-04-23:
already added it

Haxhi03, 2020-04-23:
I mean if it is possible to add a function that i ask to alexa the temperature.

kakopappa, 2020-04-24:
> please help me

https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino

Example read and send temperature to server

Haxhi03, 2020-04-24:
I use that but alexa doesn't read my temperature

kakopappa, 2020-04-24:
not sure. try switching to sinric.pro