kakopappa/sinric #417: Alexa do not change input to TV in PT-BR

lucasgiusti, 2020-05-01:
Hello,
I'm using Alexa in Portuguese / Brazil and I'm having trouble changing inputs.
When I say: Alexa, change the input of the TV to HDMI, it works correctly,
But when I say: Alexa, change the input from TV to TV, she replies “I don't know how to set the TV to this configuration”.

If possible, could you include example sentences for each Command?
For example:
Alexa, increase the volume of the tv.
Alexa, change the input of the tv to HDMI.

Tks,

kakopappa, 2020-05-21:
sorry about the late response. 

According to https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-inputcontroller.html

only these are supported

AUX 1, AUX 2, AUX 3, AUX 4, AUX 5, AUX 6, AUX 7, BLURAY, CABLE, CD, COAX 1, COAX 2, COMPOSITE 1, DVD, GAME, HD RADIO, HDMI 1, HDMI 2, HDMI 3, HDMI 4, HDMI 5, HDMI 6, HDMI 7, HDMI 8, HDMI 9, HDMI 10, HDMI ARC, INPUT 1, INPUT 2, INPUT 3, INPUT 4, INPUT 5, INPUT 6, INPUT 7, INPUT 8, INPUT 9, INPUT 10, IPOD, LINE 1, LINE 2, LINE 3, LINE 4, LINE 5, LINE 6, LINE 7, MEDIA PLAYER, OPTICAL 1, OPTICAL 2, PHONO, PLAYSTATION, PLAYSTATION 3, PLAYSTATION 4, SATELLITE, SMARTCAST, TUNER, TV, USB DAC, VIDEO 1, VIDEO 2, VIDEO 3, XBOX

lucasgiusti, 2020-05-21:
Hi,
I was successful with the following sentence (in portuguese, brazil):
Alexa, altere a entrada da "TV" (device) para "Entrada TV".
Tks,