kakopappa/sinric #418: (Sinric amazon switch config and flash) Someone new to this with a lot of errors showing up

mrTSilvers, 2020-05-10:
As I have very little experience with this sort of thing i could be doing something terribly wrong or overlooking something extremely simple, but from following the instructions on the quickstart guide and the instructions on "https://github.com/sinricpro/esp8266-esp32-sdk" To use Arduino IDE to config and flash the code to a cheap Wemos D1 mini to control a relay. I'm still getting a lot of errors when I click upload, the first being no matching function for call to 'SinricProSwitch::onPowerState, and the list goes on. Its probably overkill but I will post the error log:

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\Switch.ino: In function 'void setupSinricPro()':

Switch:108:37: error: no matching function for call to 'SinricProSwitch::onPowerState(<unresolved overloaded function type>)'

   mySwitch.onPowerState(onPowerState);

                                     ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\Switch.ino:108:37: note: candidate is:

In file included from C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProSwitch.h:11:0,

                 from C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\Switch.ino:35:

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note: virtual void SinricProDevice::onPowerState(SinricProDevice::PowerStateCallback)

 void SinricProDevice::onPowerState(PowerStateCallback cb) { 

      ^

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 'SinricProDevice::PowerStateCallback {aka std::function<bool(const String&, bool&)>}'

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_advance.ino: In function 'void setupWiFi()':

MultiSwitch_advance:146:6: error: redefinition of 'void setupWiFi()'

 void setupWiFi()

      ^

Switch:91:6: error: 'void setupWiFi()' previously defined here

 void setupWiFi() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_advance.ino: In function 'void setupSinricPro()':

MultiSwitch_advance:160:6: error: redefinition of 'void setupSinricPro()'

 void setupSinricPro()

      ^

Switch:103:6: error: 'void setupSinricPro()' previously defined here

 void setupSinricPro() {

      ^

MultiSwitch_advance:166:39: error: no matching function for call to 'SinricProSwitch::onPowerState(<unresolved overloaded function type>)'

     mySwitch.onPowerState(onPowerState);

                                       ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_advance.ino:166:39: note: candidate is:

In file included from C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProSwitch.h:11:0,

                 from C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\Switch.ino:35:

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note: virtual void SinricProDevice::onPowerState(SinricProDevice::PowerStateCallback)

 void SinricProDevice::onPowerState(PowerStateCallback cb) { 

      ^

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 'SinricProDevice::PowerStateCallback {aka std::function<bool(const String&, bool&)>}'

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_advance.ino: In function 'void setup()':

MultiSwitch_advance:173:6: error: redefinition of 'void setup()'

 void setup()

      ^

Switch:117:6: error: 'void setup()' previously defined here

 void setup() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_advance.ino: In function 'void loop()':

MultiSwitch_advance:182:6: error: redefinition of 'void loop()'

 void loop()

      ^

Switch:127:6: error: 'void loop()' previously defined here

 void loop() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_beginner.ino: In function 'void setupWiFi()':

MultiSwitch_beginner:70:6: error: redefinition of 'void setupWiFi()'

 void setupWiFi() {

      ^

Switch:91:6: error: 'void setupWiFi()' previously defined here

 void setupWiFi() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_beginner.ino: In function 'void setupSinricPro()':

MultiSwitch_beginner:83:6: error: redefinition of 'void setupSinricPro()'

 void setupSinricPro() {

      ^

Switch:103:6: error: 'void setupSinricPro()' previously defined here

 void setupSinricPro() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_beginner.ino: In function 'void setup()':

MultiSwitch_beginner:104:6: error: redefinition of 'void setup()'

 void setup() {

      ^

Switch:117:6: error: 'void setup()' previously defined here

 void setup() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_beginner.ino: In function 'void loop()':

MultiSwitch_beginner:110:6: error: redefinition of 'void loop()'

 void loop() {

      ^

Switch:127:6: error: 'void loop()' previously defined here

 void loop() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino: In function 'bool onPowerState(const String&, bool&)':

MultiSwitch_intermediate:50:6: error: redefinition of 'bool onPowerState(const String&, bool&)'

 bool onPowerState(const String &deviceId, bool &state) {

      ^

Switch:63:6: error: 'bool onPowerState(const String&, bool&)' previously defined here

 bool onPowerState(const String &deviceId, bool &state) {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino: In function 'void setupWiFi()':

MultiSwitch_intermediate:60:6: error: redefinition of 'void setupWiFi()'

 void setupWiFi() {

      ^

Switch:91:6: error: 'void setupWiFi()' previously defined here

 void setupWiFi() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino: In function 'void setupSinricPro()':

MultiSwitch_intermediate:73:6: error: redefinition of 'void setupSinricPro()'

 void setupSinricPro() {

      ^

Switch:103:6: error: 'void setupSinricPro()' previously defined here

 void setupSinricPro() {

      ^

MultiSwitch_intermediate:77:39: error: no matching function for call to 'SinricProSwitch::onPowerState(<unresolved overloaded function type>)'

     mySwitch.onPowerState(onPowerState);

                                       ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino:77:39: note: candidate is:

In file included from C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProSwitch.h:11:0,

                 from C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\Switch.ino:35:

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note: virtual void SinricProDevice::onPowerState(SinricProDevice::PowerStateCallback)

 void SinricProDevice::onPowerState(PowerStateCallback cb) { 

      ^

C:\Users\zas01\Documents\Arduino\libraries\SinricPro\src/SinricProDevice.h:145:6: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 'SinricProDevice::PowerStateCallback {aka std::function<bool(const String&, bool&)>}'

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino: In function 'void setup()':

MultiSwitch_intermediate:87:6: error: redefinition of 'void setup()'

 void setup() {

      ^

Switch:117:6: error: 'void setup()' previously defined here

 void setup() {

      ^

C:\Users\zas01\AppData\Local\Temp\arduino_modified_sketch_877423\MultiSwitch_intermediate.ino: In function 'void loop()':

MultiSwitch_intermediate:93:6: error: redefinition of 'void loop()'

 void loop() {

      ^

Switch:127:6: error: 'void loop()' previously defined here

 void loop() {

      ^

Multiple libraries were found for "SinricPro.h"
 Used: C:\Users\zas01\Documents\Arduino\libraries\SinricPro
 Not used: C:\Users\zas01\Documents\Arduino\libraries\esp8266-esp32-sdk-master
exit status 1
no matching function for call to 'SinricProSwitch::onPowerState(<unresolved overloaded function type>)'

kakopappa, 2020-05-10:
> note: no known conversion for argument 1 from '' to

Multiple libraries were found for "SinricPro.h"
Used: C:\Users\zas01\Documents\Arduino\libraries\SinricPro

Not used:
C:\Users\zas01\Documents\Arduino\libraries\esp8266-esp32-sdk-master
exit status 1
no matching function for call to 'SinricProSwitch::onPowerState()'

Try deleting  C:\Users\zas01\Documents\Arduino\libraries\SinricPro

mrTSilvers, 2020-05-10:
Thanks. 
Yeah, that took away the multiple libraries problem. However, the rest are still there. A lot of them seem to be in relation to no matching functions, which I cannot understand unless I'm missing a library, but the instructions didn't mention any other libraries.

kakopappa, 2020-05-10:
I am not sure why as well..

Can you try this

1. Create a new sketch in Arduino. Make sure you can compile it

2. Copy the code from switch sketch to the above sketch and compile

3. Change the details and compile