kakopappa/sinric #420: Report state for GH

murilobaliego, 2020-05-21:
Hi, I'm using thermostat and I'm able to update the server with DHT values. However to update the mobile google app I need to close the screen of Thermostat and open it again, it does not automatically update when Thermostat screen is opened.
I saw that this update could be possible using the report state, which "forces" the QUERY form server.
https://developers.google.com/assistant/smarthome/develop/report-state

How can we have it in sirinc? Is Report State already implemented?
Thanks in advance

kakopappa, 2020-05-21:
Report state is already implemented

murilobaliego, 2020-05-21:
Thanks @kakopappa , so How can I get the app updated without closing and opening?