kakopappa/sinric #422: Add IR signals to TV sketch in Sinric Pro

Haxhi03, 2020-05-30:
Anyone knows how to add IR signals to TV sketch in Sinric Pro.
I tried with sinric and it works, but I have my temperature sensor in Sinric Pro so I wanted to use the Temp sensor and the TV in Sinric Pro.

kakopappa, 2020-05-30:
Have you tried the TV example sketch for Sinric Pro? You should be able to
copy the code from existing sketch into this and get it to work without any
issues

https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/examples/TV/TV.ino

Haxhi03, 2020-05-31:
i use the tv example of sinric pro and i want to add the ir signals

sivar2311, 2020-05-31:
You have to implent the IR function in all necessary callback functions (onPowerState, onSkipChannel etc etc...).
This highly depends to your TV model (what IR codes are necessary) and which IR library you are using.

Do you have a working example for sending IR commands to your TV (independent to SinricPro) ?

Haxhi03, 2020-05-31:
on sinric it works but i want to transfer the ir signals to sinric prp

sivar2311, 2020-05-31:
Sending IR signals is independent to the used Sinric version. 
Do you have a specific question? Maybe you can show your code and comment the lines where you have problems.

Haxhi03, 2020-06-04:
i use the example code for tv of sinric pro and i wanted to put ir signals to that code, but i don't know how to do that

sivar2311, 2020-06-04:
First of all you need to know how to send IR codes and what codes you TV needs.
The [Arduino-IRemote](https://github.com/z3t0/Arduino-IRremote) library might be the place to start about sending IR-Codes.

Then put your code into each callback function right after the `Serial.printf`.
Example:
```
bool onMute(const String &deviceId, bool &mute) {
  Serial.printf("TV volume is %s\r\n", mute?"muted":"unmuted");

// Put your ir-send code for mute / unmute (dependent to the mute variable) here

  tvMuted = mute; // set tvMuted state
  return true;
}
```

Haxhi03, 2020-06-04:
ok but for channels an for volume?

Haxhi03, 2020-06-04:
#include <Arduino.h>
#ifdef ESP8266 
       #include <ESP8266WiFi.h>
#endif 
#ifdef ESP32   
       #include <WiFi.h>
#endif

#include "SinricPro.h"
#include "SinricProTV.h"

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRutils.h>

#define WIFI_SSID           ""    
#define WIFI_PASS           ""
#define APP_KEY             ""      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET          ""   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"
#define BAUD_RATE           115200                // Change baudrate to your need (used for serial monitor)

#define TV_ID             ""   

#define IR_LED 4

IRsend irsend(IR_LED);

uint16_t TV_POWER_H[68] = {4500,4450,550,1700,550,1650,600,1650,550,550,550,550,550,550,550,550,550,600,550,1650,550,1700,550,1650,550,550,550,550,550,600,550,550,550,550,600,500,550,1700,550,550,550,550,550,550,550,600,550,550,550,550,600,1600,600,550,550,1650,550,1700,550,1650,550,1650,600,1650,550,1650,600,};
uint16_t TV_INPUT_H[68] = {4500,4450,550,1700,500,1700,550,1650,550,600,550,550,550,550,550,550,550,600,500,1700,550,1700,500,1700,550,550,550,600,500,600,500,600,550,550,550,1700,500,600,550,550,500,600,550,550,550,600,500,600,550,550,550,550,550,1700,550,1650,550,1700,500,1700,550,1650,550,1700,550,1650,550,};
uint16_t TV_VOLUP_H[68] = {4500,4450,550,1650,550,1700,550,1650,550,550,550,600,550,550,550,550,550,550,550,1700,550,1650,550,1700,500,600,550,550,550,550,550,600,500,600,550,1650,550,1700,500,1700,550,550,550,600,500,600,550,550,550,550,550,550,550,600,500,600,550,1650,550,1700,550,1650,550,1700,550,1650,550,};
uint16_t TV_VOLDOWN_H[68] = {4500,4450,550,1700,500,1700,550,1650,550,600,550,550,500,600,550,550,550,600,500,1700,550,1700,500,1700,550,550,550,550,550,600,500,600,550,550,550,1650,550,1700,550,550,550,1700,500,600,550,550,550,550,550,550,550,600,550,550,550,1650,550,600,500,1700,550,1700,500,1700,550,1650,550,};
uint16_t TV_OK_H[68] = {4500,4450,550,1700,550,1650,550,1700,550,550,550,550,550,550,550,600,550,550,550,1650,550,1700,550,1650,550,600,500,600,550,550,550,600,500,600,550,550,550,550,550,550,550,1650,550,600,550,1650,550,1650,600,550,550,1650,600,1650,550,1650,600,550,550,1650,550,550,550,600,550,1650,550,};
uint16_t TV_MUTE_H[68] = {4500,4450,550,1650,550,1700,550,1650,550,600,550,550,550,550,550,600,500,600,500,1700,550,1650,550,1700,550,550,550,600,500,550,550,600,550,550,550,1650,600,1650,550,1650,550,1700,550,550,550,600,500,600,550,550,550,550,550,550,550,600,500,600,550,1650,550,1650,600,1650,550,1650,550,};
uint16_t TV_UP_H[68] = {4500,4450,550,1650,600,1650,550,1650,600,550,550,550,550,550,550,550,550,600,550,1650,550,1650,600,1600,600,550,550,550,550,600,550,550,550,550,550,550,550,1650,550,600,550,550,550,1650,600,550,550,550,550,550,550,1650,600,550,550,1650,550,1650,600,550,550,1650,600,1600,600,1650,550,};
uint16_t TV_DOWN_H[68] = {4500,4450,550,1650,600,1650,550,1650,550,600,550,550,550,550,550,550,550,600,550,1650,550,1650,550,1700,550,550,550,550,550,600,550,550,550,550,550,550,550,550,550,600,550,550,550,1650,550,600,550,550,550,550,550,1650,600,1650,550,1650,550,1700,550,550,550,1650,600,1650,550,1650,550,};
uint16_t TV_1[68] = {4500,4450,550,1700,550,1650,550,1650,550,600,550,550,550,550,550,550,550,600,550,1650,550,1650,550,1700,550,550,550,550,550,600,550,550,550,550,550,550,600,500,550,1700,550,550,550,550,550,600,550,550,550,550,550,1650,550,1700,550,550,550,1700,550,1650,550,1650,600,1650,550,1650,550,};
uint16_t TV_2[68] = {4450,4500,550,1700,500,1700,500,1700,550,550,550,600,500,600,500,600,550,550,550,1700,500,1700,500,1700,550,600,550,550,500,600,550,550,550,600,500,1700,550,600,500,1700,550,550,550,550,550,550,550,600,500,600,550,550,550,1700,500,600,500,1750,500,1700,500,1700,550,1700,500,1700,550,};
uint16_t TV_3[68] = {4450,4450,550,1700,550,1700,500,1700,550,550,550,550,550,600,500,600,500,600,550,1650,550,1700,550,1650,550,550,550,600,550,550,550,550,550,550,550,600,500,1700,550,1650,550,600,550,550,550,550,550,550,550,600,500,1700,500,600,550,550,550,1700,550,1650,550,1700,550,1650,550,1700,550,};
uint16_t TV_4[68] = {4450,4500,500,1700,550,1700,500,1700,550,550,550,600,500,600,500,600,500,600,550,1700,500,1700,550,1700,500,600,500,600,500,600,550,600,500,600,500,600,500,600,550,600,500,1700,500,600,500,600,550,600,450,600,550,1700,550,1700,550,1650,550,550,550,1700,500,1700,550,1700,500,1700,500,};
uint16_t TV_5[68] = {4450,4500,550,1650,550,1700,500,1700,500,600,550,550,550,550,550,600,500,600,550,1700,500,1700,500,1750,500,600,500,600,500,600,550,550,550,600,500,1700,550,550,550,550,550,1700,500,600,550,550,550,600,500,600,500,600,500,1750,500,1700,500,600,550,1700,500,1700,550,1650,550,1700,500,};
uint16_t TV_6[68] = {4450,4500,500,1700,500,1750,500,1700,500,600,550,600,500,600,500,600,500,600,500,1750,500,1700,500,1750,500,600,500,600,500,600,500,600,550,600,500,600,500,1700,550,550,550,1700,500,600,500,600,550,600,500,550,550,1700,550,600,500,1700,500,600,500,1750,500,1700,500,1700,550,1700,500,};
uint16_t TV_7[68] = {4450,4500,550,1650,550,1700,550,1650,550,550,550,600,550,550,550,550,550,550,550,1700,550,1650,550,1700,500,600,550,550,550,550,550,600,500,600,550,550,550,550,550,1700,500,1700,550,550,550,550,550,600,550,550,550,1650,550,1700,550,550,550,550,550,1700,550,1650,550,1700,500,1700,550,};
[... cut ...]
  // setup SinricPro
  SinricPro.onConnected([](){ Serial.printf("Connected to SinricPro\r\n"); }); 
  SinricPro.onDisconnected([](){ Serial.printf("Disconnected from SinricPro\r\n"); });
  SinricPro.begin(APP_KEY, APP_SECRET);
}

// main setup function
void setup() {
  Serial.begin(BAUD_RATE); 
  irsend.begin();
  Serial.printf("\r\n\r\n");
  Serial.printf("%d channels configured\r\n", MAX_CHANNELS);

  setupWiFi();
  setupSinricPro();
  setupChannelNumbers();
}

void loop() {
  SinricPro.handle();
}

sivar2311, 2020-06-04:
> ok but for channels an for volume?

Please check the [documention](https://sinricpro.github.io/esp8266-esp32-sdk/class_sinric_pro_t_v.html). 
For channels use

- onChangeChannel
- onChangeChannelNumber
- onSkipChannels

For volume use
- onSetVolume
- onAdjustVolume
- onMute

sivar2311, 2020-06-04:
Ok, but now everything is clear?

Haxhi03, 2020-06-05:
no

Haxhi03, 2020-06-05:
cause i don't know how to add ir signals to volume and channel

sivar2311, 2020-06-05:
What do you mean by "add ir signals" ?
There are a lot of ir-codes defined in your sketch!
For example switch to channel number 23 the `onChangeChannelNumber` callback gets called with 23 as channel parameter. So you have to send the IR-codes for `2`, `3` and `OK`

Haxhi03, 2020-06-06:
ok thank you

Haxhi03, 2020-06-06:
i solved for channel, but the bool setVolume i can't set it.

sivar2311, 2020-06-06:
Ok, what's exact the problem with volume?

Haxhi03, 2020-06-06:
the problem is tha if i say Alexa set television volume to 20, the volume raised to 100

sivar2311, 2020-06-06:
What does the serial monitor print if you set the volume to 20%?
If it prints 20%, the problem is outside of sinric pro.

Haxhi03, 2020-06-06:
it's outside of sinric pro
can you help me if i post the code ?

sivar2311, 2020-06-06:
Because it depends very much on the TV, I think you just have to try it yourself.

kakopappa, 2020-06-06:
You have to find the correct IR code(s) some TVs use the same IR code to
increase / decrease

kakopappa, 2020-06-11:
@laxmru @sivar2311 Websocket 2.1.4 seems not compatible anymore with ESP32 latest core. Updating to 2.2.0 seems to fix the compiling errors

laxmru, 2020-06-12:
> To be able to help you we need more information! Please provide us at least the error log and information about
> - the version of the installed webSocket library
> - the platform (ESP8266 / ESP32)?
> - the IDE used
> - the Espressif SDK version used
> - the used Arduino Core version
> 
> Since it is related to SinricPro, please open a [new issue in the SinricPro repository](https://github.com/sinricpro/esp8266-esp32-sdk/issues) and provide the above information there.
> 

I have solved this problem using Nodemcu V1 board. Earlier I was using Arduino UNO with ESP8266 module.
The new problem which am facing is the temperature value gets updated at the time of uploading sketch. For example, I get 31 ° as soon as I upload the sketch and then it doesn't change. It doesn't refresh after that. If I restart the Nodemcu it changes to 1 °.
The error message I get in serial monito is "DHT reading failed"

laxmru, 2020-06-16:
> the problem is tha if i say Alexa set television volume to 20, the volume raised to 100

It could be because it has maximum 20 levels of volume.

sivar2311, 2020-06-16:
You need to map the percentage volume level (0..100) to real value levels (0..20).
Use the [arduino map function](https://www.arduino.cc/reference/en/language/functions/math/map/) for this.

j8prakash, 2020-08-28:
> Anyone knows how to add IR signals to TV sketch in Sinric Pro.
> I tried with sinric and it works, but I have my temperature sensor in Sinric Pro so I wanted to use the Temp sensor and the TV in Sinric Pro.

have you managed to add ir signal to sketch sinricpro.

j8prakash, 2020-08-29:
does anyone has Sinric pro TV sketch with ir codes . i have working sketch for sinric but i want to upgrade to Sinric pro. please can someone help.

Joshthynne, 2021-02-02:
I am currently working on a youtube tutorial for exactly this. Planning on dropping it in a couple of weeks. https://www.youtube.com/channel/UCQlMVtTTtNYMCdhDZO5dGuw

j8prakash, 2021-02-04:
Thanks boss

Joshthynne, 2021-02-17:
A good, easy to follow tutorial for combining all of your IR remotes into one simple board, which integrates into Alexa via SinricPro. https://youtu.be/nh5adggIJYo