kakopappa/sinric #427: Sinric Google Home Integration not working

EkanshdeepGupta, 2020-06-12:
Hi,
I followed the tutorial at https://github.com/kakopappa/sinric/wiki/How-to-use-Google-Home, clicked on Test and everything. I successfully added the devices to Google Home App as well.

Upon sending command, such as "turn fan off", google responds like "sure" (meaning it is detecting device), however the ESP8266 board does not respond.

It is responding when I control it using the Sinric App from the playstore (https://play.google.com/store/apps/details?id=com.sinric). However, if I change Command Format from Alexa to Google Home, there is no response from the app as well.

I really want Google Home Integration since I have a Google Home device. Can you please help me troubleshoot why this is not working?

The device is being detected by the Google Home, but any commands don't get executed. On the app also, commands get executed when Command Format is set to Alexa, but not when Command Format is set to Google Home.

Thank you so much.

kakopappa, 2020-06-13:
Can you give Sinric Pro a try ? https://sinric.pro

You can use IFTTT to send commands from GH to Sinric Pro via API. It works
well.

EkanshdeepGupta, 2020-06-13:
Hi,
Thanks for your suggestion. I just made a Sinric Pro account. Can you guide me a bit about implementing Sinric Pro? Will the same code as this Sinric work for my ESP8266? 

I looked at https://help.sinric.pro/pages/quickstart.html, however the ESP8266 demo code example  on that page is showing Not Found.

kakopappa, 2020-06-13:
1. install the sinricpro sdk from the library manager in Arduino

2. Get the sample code from

https://github.com/sinricpro/esp8266-esp32-sdk

3. Change the credentials and flash.

4. Try turning on or off using the portal.

5. Follow the tutorial to setup IFTTT

Let me know what you think and experience to make it better

EkanshdeepGupta, 2020-06-14:
Hi kakopappa,

I was having some trouble with the example codes given. Lots of code examples don't have any reference to the output pin on the ESP8266 board. For example the Light/Light.ino and Fan/Fan.ino scripts don't have a parameter defined for output pin (as far as I can tell). So how is the ESP8266 board supposed to know how to turn off the respective device if I don't give it an output pin?

I am a bit confused about the setup, so I was hoping you could shed some light.

I was wondering in general if the same code works for Sinric Pro as for this version of Sinric (by providing the basic functionality, for switches, say), so maybe I don't have to figure everything out again? Please let me know if that is the case. And please help me understand my queries above.

Thanks!