kakopappa/sinric #435: (Sinric) Can't log in! :(

DaniloFloyd, 2020-08-02:
Hello everyone!!

I've make a new account, trying to link it to my Amazon/Alexa, Skills, etç. It´s not accepting my key no matter what, neither on the app or the amazon sinric skill page.

Help please?

kakopappa, 2020-08-03:
Not sure what you mean by “. It´s not accepting my key no matter what,”

You are suppose to enter your email/password for sinric here.

Make sure skill name is Sinric for sinric.com and Sinric Pro for sinric.pro

DaniloFloyd, 2020-08-03:
> Not sure what you mean by “. It´s not accepting my key no matter what,”
> 
> You are suppose to enter your email/password for sinric here.
> 
> Make sure skill name is Sinric for sinric.com and Sinric Pro for sinric.pro
> …

((("Make sure skill name is Sinric for sinric.com and Sinric Pro for sinric.pro")))

Oh sir, thank you very much! Yes, i was confusing those two...how stupid!
Thanks again and have a great week!

closed 2020-08-03 by DaniloFloyd