kakopappa/sinric #436: not getting response from sinric in esp 32

muhammad8khan, 2020-08-09:


muhammad8khan, 2020-08-09:
i am using sinric from past 5 months in esp 32 and everything was good from from some days my esp not giving any response and getting discontented continuously( as u can see in picture ) i am using below code.
[image]
`
#include <Arduino.h>
#include <WiFi.h>   // http://arduino.esp8266.com/stable/package_esp8266com_index.json
#include <WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

int relay1 = 32;
int relay2 = 33;
int relay3 = 25;
int relay4 = 26;
int relay5 = 27;
int relay6 = 14;
int relay7 = 23;
int relay8 = 22;
int LED = 2;

WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "*************************************" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "*************" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "***********" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5f2fd09bad7a48327f3997b6") // Device ID of 1 device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(relay1,LOW); 
  } 
  else if (deviceId == "5f2fd0acad7a48327f3997b8") // Device ID of 2 device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(relay2,LOW);
  }
    else if (deviceId == "5f2fd0b8ad7a48327f3997bc") // Device ID of 3 device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(relay3,LOW);
  }
   else if (deviceId == "5f2fd0c7ad7a48327f3997c0") // Device ID of 4 device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(relay4,LOW);
  }
   else if (deviceId == "5f2fd0d7ad7a48327f3997c2") // Device ID of 5 device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(relay5,LOW);
  }
   else if (deviceId == "5f2fd0e4ad7a48327f3997c6") // Device ID of 6 device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
[... 167 lines cut ...]

  
  webSocket.begin("iot.sinric.com", 80, "/");

  
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);  
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
     
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}`

kakopappa, 2020-08-09:
I have restarted the server. Check now. You should consider moving to
Sinric Pro