kakopappa/sinric #437: UNINTENDED TURNING ON OF RELAYS ON START UP

ajarsco, 2020-08-11:
When ever my powersupply goes off and comes back up.. all the relays get switched on and not to the previous state .. how can i correct this..
Kindly reply fast..

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "" // TODO: Change to your Wifi network Password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// Switch
const int BUTTON1_PIN = 15;
const int BUTTON2_PIN = 5;
const int BUTTON3_PIN = 4;
const int BUTTON4_PIN = 0;

//Relays
const int RELAY1_PIN = 2;
const int RELAY2_PIN = 14;
const int RELAY3_PIN = 12;
const int RELAY4_PIN = 13;

//Status LEDs
const int LED1 = 16;
const int LED2 = 3;
const int LED3 = 1;

String device_ID_1 = "";
String device_ID_2 = "";
String device_ID_3 = "";
String device_ID_4 = "";

ButtonConfig config1;
AceButton button1(&config1);
ButtonConfig config2;
AceButton button2(&config2);
ButtonConfig config3;
AceButton button3(&config3);
ButtonConfig config4;
AceButton button4(&config4);

void handleEvent1(AceButton*, uint8_t, uint8_t);
void handleEvent2(AceButton*, uint8_t, uint8_t);
void handleEvent3(AceButton*, uint8_t, uint8_t);
void handleEvent4(AceButton*, uint8_t, uint8_t);

void setPowerStateOnServer(String deviceId, String value);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == device_ID_1) // Device ID of first device
  {
    Serial.print("Turn on device id: 1");
    Serial.println(deviceId);
[... 289 lines cut ...]
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_3, "OFF");
      digitalWrite(RELAY3_PIN, HIGH);
      break;
  }
}

void button4Handler(AceButton* button, uint8_t eventType, uint8_t buttonState) 
{
  Serial.println("EVENT4");
  switch (eventType) 
  {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer(device_ID_4, "ON");
      digitalWrite(RELAY4_PIN, LOW);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_4, "OFF");
      digitalWrite(RELAY4_PIN, HIGH);
      break;
  }
}

sivar2311, 2020-08-11:
I think you have 2 basic options here:

1) use EEPROM (needs to be implemented by you):
You can store the state in eeprom on each state-change.
Then in setup() you can read the last known state from eeprom.

2) Switch over to SinricPro
Make use of `SinricPro.restoreDeviceStates(true)` functionality

ajarsco, 2020-08-11:
sir, can we use SPIFFS. Or is it fine to eeprom better

sivar2311, 2020-08-11:
Short answer: Yes, but...
SPIFFS is depracted. If you want to use a file-system for this (little task) you should use LittleFS.

Long answer: Yes but...
SPIFFS is depracted. If you want to use a file-system for this (little task) you should use LittleFS.
Do you really want to use a complete file-system just for storing 4 Bits of information? (4 Relais = 4x true or false = 4 Bits of information needed)
- It's a waste of of valuable space
- It produces much more overhead (open file, read from file, write to file...) which ends up in much more lines of code

sivar2311, 2020-08-11:
Examples: click [here](https://www.google.com/search?q=Arduino+EEPROM+example)

You don't need an example for Sinric. It's a general thing, which can be used for many purposes.

ajarsco, 2020-08-11:
ok sir, then how do we solve this issue. can you give me a simpler solution for this

sivar2311, 2020-08-11:
Simplest solution is: Switch over to SinricPro. The SDK supports restoring the last known state (server sided).
Otherwise, choose one of the other suggestet solutions (EEPROM / FileSystem) which is the easiest for you.

ajarsco, 2020-08-11:
what if I use Sinric pro, that's only the server side right? But how it will work there can u please show me a sample code example with Sinric pro, so that I can switch over to that

sivar2311, 2020-08-11:
https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/Switch

ajarsco, 2020-08-11:
ok sir let me check

ajarsco, 2020-08-11:
sir, will this be ok by the SinricPro.restoreDeviceStates(true) code only

ajarsco, 2020-08-11:
without any changes in local storage methods, and using the server only can we get restore the device to normal

> sir, will this be ok by the SinricPro.restoreDeviceStates(true) code only

sivar2311, 2020-08-11:
Please see [full-documentation](https://sinricpro.github.io/esp8266-esp32-sdk/class_sinric_pro_class.html#a27a9bb7f1e8bde0f39398649493b6f93) about restoreDeviceStates()

What do you mean with "device to normal" ??

When restoreDeviceStates is enabled (true):
After connection have been established the server will send the last known states for each device as a normal action.
So no additonal code here for you, except this one line: ` SinricPro.restoreDeviceStates(true)`

kakopappa, 2020-08-11:
When you call this methods the server will send the last known status in
the server to client.

ajarsco, 2020-08-11:
ok

ajarsco, 2020-08-11:
can you give me code for controlling 4 relays and with eeprom. to store data locally

ajarsco, 2020-08-11:
kindly reply sir so that I can do with this for now

ajarsco, 2020-08-11:
> kindly reply sir so that I can do with this for now

I am a beginner so I am not able to write a code from scratch for eeprom

sivar2311, 2020-08-11:
Everything you need to know is inside the git description, full documentation and the provided examples.

We offer you everything so that you can realize your projects. 
But the work to realize your projects you have to do yourself.

I am willing to help you, if you have **specific** question about SinricPro. 
Start with a simple sketch using sinric pro and the provided simple switch examples.
Then add more and more code to your project.

About your Edit while i was writing the lines above:
As a beginner: Learning and understanding arduino is the first step.
For a beginner i recommend to start with much simpler projects than home automation - regardless which home automation system you're using.
Analogy: You have to learn to walk before you can run a marathon.

ajarsco, 2020-08-11:
sorry sir, I will do it sir

ajarsco, 2020-08-11:
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>
#include <EEPROM.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "" // TODO: Change to your Wifi network Password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 
//#define DEBOUNCE_DELAY 3000 // 3 seconds

uint64_t heartbeatTimestamp = 0;
//uint64_t DelayTimer = 0;

bool isConnected = false;

bool Device1Status = false;
bool Device2Status = false;
bool Device3Status = false;
bool Device4Status = false;

// Switch
const int BUTTON1_PIN = 15;
const int BUTTON2_PIN = 5;
const int BUTTON3_PIN = 4;
const int BUTTON4_PIN = 0;

//Relays
const int RELAY1_PIN = 2;
const int RELAY2_PIN = 14;
const int RELAY3_PIN = 12;
const int RELAY4_PIN = 13;

//Status LEDs
const int LED1 = 16;
const int LED2 = 3;
const int LED3 = 1;

String device_ID_1 = "";
String device_ID_2 = "";
String device_ID_3 = "";
String device_ID_4 = "";

ButtonConfig config1;
AceButton button1(&config1);
ButtonConfig config2;
AceButton button2(&config2);
ButtonConfig config3;
AceButton button3(&config3);
ButtonConfig config4;
AceButton button4(&config4);

void handleEvent1(AceButton*, uint8_t, uint8_t);
void handleEvent2(AceButton*, uint8_t, uint8_t);
void handleEvent3(AceButton*, uint8_t, uint8_t);
void handleEvent4(AceButton*, uint8_t, uint8_t);

void setPowerStateOnServer(String deviceId, String value);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here
[... 364 lines cut ...]
      break;
      EEPROM.commit();
  }
   
}

void button4Handler(AceButton* button, uint8_t eventType, uint8_t buttonState) {
  Serial.println("EVENT4");
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer(device_ID_4, "ON");
      digitalWrite(RELAY4_PIN, LOW);
      EEPROM.write(3, 1);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_4, "OFF");
      digitalWrite(RELAY4_PIN, HIGH);
      EEPROM.write(3, 0);
      break;
      EEPROM.commit();
  }
   
}

ajarsco, 2020-08-11:
SIR please help, the old state is stored and working only for relay 2 but not happening for others

ajarsco, 2020-08-11:
please help me sir

sivar2311, 2020-08-11:
What i would do is, adding some additional Serial.print statements for locating the issue (to check what happened and did it work like you have expected).

Example:
In front of each EEPROM write, print it to the serial what you writing to the EEPROM.
After each EEPROM write, read the value from the EEPROM and print it to the serial.
Then do a test, watch the serial monitor.

Doing this way, you will find the issue by yourself.

Btw:
Your code is hard to read. Use the "insert code" functionality or better, upload your code to gist and post a link here.

ajarsco, 2020-08-12:
`
#include <EEPROM.h>
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "" // TODO: Change to your Wifi network Password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// Switch
const int BUTTON1_PIN = 15;
const int BUTTON2_PIN = 5;
const int BUTTON3_PIN = 4;
const int BUTTON4_PIN = 0;

//Relays
const int RELAY1_PIN = 2;
const int RELAY2_PIN = 14;
const int RELAY3_PIN = 12;
const int RELAY4_PIN = 13;

//Status LEDs
const int LED1 = 16;
const int LED2 = 3;
const int LED3 = 1;

String device_ID_1 = "";
String device_ID_2 = "";
String device_ID_3 = "";
String device_ID_4 = "";

ButtonConfig config1;
AceButton button1(&config1);
ButtonConfig config2;
AceButton button2(&config2);
ButtonConfig config3;
AceButton button3(&config3);
ButtonConfig config4;
AceButton button4(&config4);

void handleEvent1(AceButton*, uint8_t, uint8_t);
void handleEvent2(AceButton*, uint8_t, uint8_t);
void handleEvent3(AceButton*, uint8_t, uint8_t);
void handleEvent4(AceButton*, uint8_t, uint8_t);

void setPowerStateOnServer(String deviceId, String value);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  bool ShouldCommit = true;
  
  if (deviceId == device_ID_1) // Device ID of first device
  {
    Serial.print("Turn on device id: 1");
[... 334 lines cut ...]
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_3, "OFF");
      digitalWrite(RELAY3_PIN, HIGH);
      break;
  }
}

void button4Handler(AceButton* button, uint8_t eventType, uint8_t buttonState) 
{
  Serial.println("EVENT4");
  switch (eventType) 
  {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer(device_ID_4, "ON");
      digitalWrite(RELAY4_PIN, LOW);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_4, "OFF");
      digitalWrite(RELAY4_PIN, HIGH);
      break;
  }
}`

ajarsco, 2020-08-12:
thank you sir, kindly check this code that I have added in insert code method for your better understanding.
hope you will find this easier. 
For certain relays the eeprom works fine and for others It doesn't. also, I think this not at all a reliable one. frequent power outages cannot handle this

sivar2311, 2020-08-12:
Seems it didn't work. Code looks still the same.
You didn't add anything  to debug (as i mentioned before) ?
Just looking at the code won't help here. You have to debug as i have described [here](https://github.com/kakopappa/sinric/issues/437#issuecomment-672147280)

sivar2311, 2020-08-12:
I did a small EEPROM test [here](https://gist.github.com/sivar2311/4627e83eeb63602e4602def958b4fac7)
You can use it to check if your EEPROM works correctly.
The sketch initializes the EEPROM at first run. Then EEPROM address 1..5 will be incremented at each boot up (to a maximum of 19...then roll over)

ajarsco, 2020-08-12:
ok sir let me check this and debug it, and I will let you know

sivar2311, 2020-08-13:
Any news here?

ajarsco, 2020-08-13:
yes sir I am in an emergency now, I will do it faster

sivar2311, 2020-08-13:
It's okay.
I only ask because you seemed to be in a hurry and now there's no answer. 
But take your time. Emergencies come first!

ajarsco, 2020-08-13:
yes sir I will do it soon

ajarsco, 2020-11-15:
everything working fine now! but there is issues in reconnecting to the wifi in between..? 

I have already used WiFi.setAutoReconnect(true) 

 why this is happening like this... this issue will remain same if we use sinric.pro also!!

Kindly help me in correcting if I have done any mistakes.

sivar2311, 2020-11-15:
Could be related to weak wifi connection.
You can log the WiFi-signal strength to serial monitor:
```
void logWiFiRSSI() {
  static unsigned long lastMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - lastMillis > 1000) {
    lastMillis = currentMillis;
    Serial.printf("WiFi Signal Strength: %d\r\n", WiFi.RSSI());
  }
}
...
void loop() {
...
  logWiFiRSSI();
...
}
```
This will print the signal strength once a second.
Value should be less than -70 dBm (Values between -30 and -70 are ok, less than -70 wifi strength is not good enough)

ajarsco, 2020-11-15:
ok sir, let me check

sivar2311, 2020-11-15:
Got any useful logs?

ajarsco, 2020-12-01:
no sir!

ajarsco, 2020-12-01:
WiFi have its maximum strength, but also the ESP fails to reconnect after long span connection(around 7-8 hours)