kakopappa/sinric #448: Esp8266 does not connect when router restarts

Mahesh1385, 2020-10-06:
Esp8266 does not connect when router restarts.
The board needs to power off and then power on to connect to the router. What may be the problem. Please help

SamueldaCostaAraujoNunes, 2020-10-06:
Inside the loop, do you check if the card is still connected to the network?

sivar2311, 2020-10-06:
I think WiFi does not reconnect automaticly.
If you restart the board, `WiFi.begin(ssid, pass)` gets called which will connect to your router.
Maybe you should try `WiFi.setAutoReconnect(true)` see [here](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.rst#setautoreconnect)

Mahesh1385, 2020-10-06:
Ok i will check the same and notify

Mahesh1385, 2020-10-06:
This is the code once check please 
int device1 = 5;
int device2 = 4;
int device3 = 0;
int device4 = 2;
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> // get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "8c3cc952-6169-4c49-910d-75876501b112" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID " " // TODO: Change to your Wifi network SSID
#define MyWifiPassword "srinika123" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
 if (deviceId == "5f6d91b7d1e9084a7085f0ee") // Device ID of device 1
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device1, LOW);
 } 
 else if (deviceId == "5f6d91ead1e9084a7085f0f2") // Device ID of device 2
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device2, LOW);
 }
 else if (deviceId == "5f6d91f7d1e9084a7085f0ff") // Device ID of device 3
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device3, LOW);
 } 
 else if (deviceId == "5f6d9202d1e9084a7085f101") // Device ID of device 4
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device4, LOW);
 }
 else 
 {
 Serial.print("Turn on for unknown device id: ");
 Serial.println(deviceId); 
 } 
}

void turnOff(String deviceId) {
 if (deviceId == "5f6d91b7d1e9084a7085f0ee") // Device ID of device 1
 { 
 Serial.print("Turn off Device ID: ");
 Serial.println(deviceId);
 digitalWrite(device1, HIGH);
 }
 else if (deviceId == "5f6d91ead1e9084a7085f0f2") // Device ID of device 2
[... 134 lines cut ...]
 root["action"] = "setPowerState";
 root["value"] = value;
 StreamString databuf;
 root.printTo(databuf);
 
 webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
 DynamicJsonBuffer jsonBuffer;
 JsonObject& root = jsonBuffer.createObject();
 root["action"] = "SetTargetTemperature";
 root["deviceId"] = deviceId;
 
 JsonObject& valueObj = root.createNestedObject("value");
 JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
 targetSetpoint["value"] = value;
 targetSetpoint["scale"] = scale;
 
 StreamString databuf;
 root.printTo(databuf);
 
 webSocket.sendTXT(databuf);
}

sivar2311, 2020-10-06:
Sorry, i dont have the time to analyze your code. But you can change to SinricPro. Our sdk reconnects websocket connection automaticly (except wifi, but for this setAutoReconnect)

Mahesh1385, 2020-10-06:
is sinric pro free to use?

kakopappa, 2020-10-06:
Yes. 3 devices are free to use. no limitations

Mahesh1385, 2020-10-06:
ok thank you for the fast response

akaith, 2020-10-22:
> This is the code once check please
> int device1 = 5;
> int device2 = 4;
> int device3 = 0;
> int device4 = 2;
> #include <Arduino.h>
> #include <ESP8266WiFi.h>
> #include <ESP8266WiFiMulti.h>
> #include [...]

I was facing the same issue, the websocket was not able to know when the connection was lost and websocket did not establish a connection again unless the board was reset. The heartbeat checking is the one which is supposed to ensure that the connection loss is detected and reconnect websocket. I discovered that websocket library has an inbuilt function which can be used to sort out this issue. You can just avoid using the manual heartbeat checking and use the following line of code within setup(), the problem should be resolved. It was resolved for me. 

// can be put just after the " webSocket.setReconnectInterval(5000); " line in the code and heartbeattimestamp part can be //deleted as it servers no purpose anymore.
webSocket.enableHeartbeat(15000, 3000, 2);

I've spent quite some time to find an issue for the same problem and i found a solution a couple of days back n did a few tests to verify that the problem was resolved. Hope this helps. Do let me know if the issue is resolved for you as well. 

Thank you

Mahesh1385, 2020-10-27:
Thanks for the response. I can not understand what you said because i don't have basic knowledge in this. Please could you say where to replace and where to delete in detail