kakopappa/sinric #449: Is there any difference in Alexa and Google Home code except the api key?

uddin232, 2020-10-06:
i have this code made for sinric to work with Alexa but i have a feeling that it might be for Alexa only and might not work with Google Home so if possible can you tell me what changes should i make in this to make it work with Google Home?

/*
  This is the code for the project called

  Alexa & Manual Controlled Home Automation using Sinric

  This code is written by Sachin Soni on 04.07.2020

  The tutorial Video for the project is uploaded on 
  our YouTube channel called "techiesms"

  Channel Link - https://www.youtube.com/techiesms

          techiesms
  explore | learn | share 
*/

#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;
WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "Your_API_Key" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "SSID" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "PASS" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// Switch
const int BUTTON1_PIN = 32;
const int BUTTON2_PIN = 35;
const int BUTTON3_PIN = 34;
const int BUTTON4_PIN = 39;

//Relays
const int RELAY1_PIN = 15;
const int RELAY2_PIN = 2;
const int RELAY3_PIN = 4;
const int RELAY4_PIN = 22;

//Status LEDs
const int LED1 = 26;
const int LED2 = 25;
const int LED3 = 27;

String device_ID_1 = "DeviceID1";
String device_ID_2 = "DeviceID2";
String device_ID_3 = "DeviceID3";
String device_ID_4 = "DeviceID4";

ButtonConfig config1;
AceButton button1(&config1);
ButtonConfig config2;
AceButton button2(&config2);
ButtonConfig config3;
AceButton button3(&config3);
ButtonConfig config4;
AceButton button4(&config4);
[... 293 lines cut ...]
      digitalWrite(RELAY3_PIN, LOW);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_3, "OFF");
      digitalWrite(RELAY3_PIN, HIGH);
      break;
  }
}

void button4Handler(AceButton* button, uint8_t eventType, uint8_t buttonState) {
  Serial.println("EVENT4");
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer(device_ID_4, "ON");
      digitalWrite(RELAY4_PIN, LOW);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_4, "OFF");
      digitalWrite(RELAY4_PIN, HIGH);
      break;
  }
}

uddin232, 2020-10-06:
i also would like to know how to convert Sinric Code to Sinric Pro code as i feel sinric is the older version and might become obsolete for Google Home as you have done with Alexa

sivar2311, 2020-10-06:
The Arduino Code for ESP is not responsible for handling alexa or google requests. 
Communication to alexa, google and other services is done server sided.
Even the SinricProSDK is only for communication between your device (ESP) and SinricPro server.
Check out the [examples](https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples) for SinricPro. You'll see it is more comfortable to use and a lot more devices-types are available. (You don't have to handle all this websocket stuff)
Unfortunately, Google Home is not (yet) natively supported (pending for aproval). 
But until aproval there are two work around for this. Best one is to use Samsung SmartThings - see [here](https://help.sinric.pro/pages/smartthings.html)

kakopappa, 2020-10-07:
In Sinric.com Alexa vs Google Home command format is different

Check the google home example

https://github.com/kakopappa/sinric/blob/master/arduino_examples/google_home_switch_example.ino

Line 91