kakopappa/sinric #456: can control lock via Sinric Dashboard, but not with SmartThings app (bug)

societyofrobots, 2020-11-12:
I got my lock device working great via the Dashboard:
https://portal.sinric.pro/dashboard

And in the SmartThings app, the device is correctly shown and indicates it's connected. It says, "unlocked".

but:
- It doesn't update it's status in the SmartThings app when I make a change via the Dashboard.
- When I click the device circle to lock or unlock in the ST app, nothing happens with the lock. The Sinric app does a popup whenever I click the circle, but the lock doesn't physically unlock.

kakopappa, 2020-11-13:
SmartThings doesn’t support unlock for security reasons. Try Sinric Pro app

societyofrobots, 2020-11-13:
Interesting. Is SmartThings an inherently insecure app? Is this also why it doesn't show 'unlocked'?