kakopappa/sinric #461: Issues

guicovred, 2021-01-05:
Hi, I'm having issues to access the website and my devices just stopped working... is there a problem?

This started to happen a few days ago.

https://sinric.pro/pt-index.html: 
[image]

kakopappa, 2021-01-05:
@guicovred  your IP has been blocked because of it's making too many HTTP requests. do you have any scripts running?

guicovred, 2021-01-05:
Well, no ... but my IP is not static, there may be some issue with my provider. I use just few devices in my home and turn they on / off only in Alexa

kakopappa, 2021-01-05:
Well, one or more devices are crashing and rebooting and making too many requests to the server. Check the attached video

https://user-images.githubusercontent.com/7474406/103673866-a5d57400-4fb0-11eb-89df-b944ef1af66e.mp4

guicovred, 2021-01-05:
This is sad. I will check, thank you very much for your attention so far.

closed 2021-05-31 by guicovred