kakopappa/sinric #468: Device not Responding

SparshRawal, 2021-02-12:
Hi I've set up my automated switches using Sinric smart skill on Alexa and now it's showing that device is not responding and also I'm not able to log in onto Sinric.com

vaibhavbhange14, 2021-02-12:
Same issue!

kakopappa, 2021-02-12:
Restarted the server. should be okey now

SparshRawal, 2021-02-12:
Still not working and receiving 502 error!
[image](https://user-images.githubusercontent.com/51254729/107755745-3be78180-6d49-11eb-9813-df4731be6d8a.png)

tonymrl, 2021-02-12:
Same issue

SparshRawal, 2021-02-12:
Working now thanks

vaibhavbhange14, 2021-02-12:
> Restarted the server. should be okey now
> […](#)

Thanks ✌️

SparshRawal, 2021-02-12:
It's gone again .!

SparshRawal, 2021-02-12:
Please fix the issue permanently

SparshRawal, 2021-02-12:
> > Restarted the server. should be okey now
> 
> > […](#)
> 
> 
> 
> 
> Thanks ✌️

Bro is it working ??

kakopappa, 2021-02-12:
> Restarted the server. should be okey now
>
> … <#m_772573894877207946_>
>
> On Fri, Feb 12, 2021 at 5:00 PM vaibhavbhange14 *@*.***> wrote: Same

Yes. had to ban few ips.

vaibhavbhange14, 2021-02-12:
> > > Restarted the server. should be okey now
> > 
> > 
> > > […](#)
> > 
> > 
> > 
> > 
> > Thanks ✌️
> 
> Bro is it working ??

Yes✌️✌️

SparshRawal, 2021-02-12:
Thanks bro working just fine

closed 2021-02-12 by SparshRawal