kakopappa/sinric #472: Problems with connecting. Exception (9)

1237780, 2021-02-16:
Hello,
I'm trying to control a light switch either with a ESP8266 ESP-12F or with a D1 Mini V3 NodeMCU.
I tried to load the example sketch "Switch" https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/Switch/Switch on both boards.
The connection with the WiFi is successful, but after that one i run into an exception.
The exact message is "Exception (9): epc1=0x40218607 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fffff8b depc=0x00000000".
After that the whole content of the stack is printed.
After this the board is connecting with the WiFi again.
A screenshot of the Serial Monitor is attached.
Since the same thing happens with both boards and other programms are running with no problem I'm clueless.
Some help would be great.

Greetings and thank you for your help in advance.
[image]

sivar2311, 2021-02-16:
Please check for latest ESP Core Version (2.7.4) (Tools / Boards / Board Manger):
[image]
and latest webSockets Version (2.3.4) Library: Tools / Manage Libraries
[image]

1237780, 2021-02-16:
It was the older ESP Core Version, which I had to downgrade in the past for my last project.
I totally forgot about that one.
A big thanks for your very fast help and it's a great thing that you are doing with Sinric.

closed 2021-02-16 by 1237780