kakopappa/sinric #474: Info about your service

The-Great-One, 2021-02-23:
Hi @kakopappa, I have been using your service for the past few years and have built an interest in IOT. I wanted to know what resources you used to build this so that i can build an IOT server like this on my own as well. I am a computer science engineer in my 3rd year and would love to know about this. Thanks

kakopappa, 2021-03-01:
Hi

[image]

This is a high-level architecture of the current sinric.com server

When you say Alexa turn on xxx. it sends the command to the Nginx Server (Load Balancer). Nginx server decides which application server that request should go to and forward the request. The application server is written using nodejs. Following modules have been used.

expresss (for web apis)
ws (web socket)
mongoDb (for database)

When the request hits the application server it looks for the device and forwards the command. That's it. If you are going to build your own server, break it down to small parts and implement it separately and connect them at the end. 

For building the smart home Alexa skill you can check here
https://developer.amazon.com/en-US/docs/alexa/smarthome/steps-to-build-a-smart-home-skill.html

The-Great-One, 2021-03-25:
Thanks for the reply. 

I also wanted to know what service are you using to manage all the devices because i tried aws iot core and it asks me to create like 3 different cert files and upload them to the nodemcu.

kakopappa, 2021-03-26:
You have to write your own server to handle the requests from alexa and others. we donot use iot core.