kakopappa/sinric #475: Facing Issue on dimming the light  --- No Physical Change

joy-sen, 2021-03-06:
Hi, 
i am beginner to this IOT platform. If any wrong, please correct me.
 
1)  For dimming the lights should i have to write any code inside the below conditions :

else if(action == "AdjustBrightness") {
            // alexa, dim lights  ==>{"deviceId":"xxx","action":"AdjustBrightness","value":-25}
       }  

else if(action == "SetBrightness") {
           //alexa, set the lights to 50% ==> {"deviceId":"xx","action":"SetBrightness","value":50}
        }

why am i asking because, though Alexa responding with "OK" but i am not getting any changes on brightness of light.
Even the serial monitor showing the proper value like ---
[webSocketEvent] get text: {"deviceId":"60430057c26766757ec0ccd4","action":"AdjustBrightness","value":-25}

can refer the below screen shot of serial monitor.

[image]

2) While giving the "dim" command the light is turning OFF.

I have tested this code by using both Bread Board and Relay. In both the cases the result is same.

should it required any hardware or any kind of dimmer circuit or any specific light ?

Please help me to suggest some solution to make it work. your quick reply will be highly appreciable.

Thanks.

sivar2311, 2021-03-06:
Have you seen the message on the sinric.com site?
This project will be discontinued as of August 31.

Switch to SinricPro (https://sinric.pro)

joy-sen, 2021-03-06:
Thanks a lot. Just now i seen the message on site.

But i have one doubt. If it is discontinued then how am i able to use the ON OFF feature using sinric? 

Only facing issue on dimming.

sivar2311, 2021-03-06:
All sinric devices will stop working after August 31.
You need to switch over to SinricPro.

For SinricPro arduino library and examples are available.