kakopappa/sinric #476: Start a Google Home Command using Sinric (not vice versa)

Inframagnetite, 2021-03-22:
Is there any workaround to do what i asked in the headline?
The applications for me would be numerous. 

High precision event timers (timing takes place in the ESP, command is issued afterwards). E.g. My Window shutters are attached to stepper drivers and open after exactly 7h and 40mins after the "good night" command. 
I want my home pod to tell me my appointments for the day right after opening the shutters without having to ask again. 

The dirtiest solution would be to open up the home pod and attach the arduino analogue out to the microphone and play tiny mp3 files. But there must be a more elegant solution. 
Probably using webhooks or sth alike. 

Any help is appreciated.

kakopappa, 2021-03-22:
I think  you can handle the good night closing blinds in the ESP  and open
the window shutters after 7h 40 mins delay later.
use this project to speak stuff on GH
https://github.com/horihiro/esp8266-google-home-notifier

Inframagnetite, 2021-03-22:
They do close and open with the esp. But after opening up, I want to issue a GH command, initiated by the ESP. 
Is this possible?