kakopappa/sinric #478: Google Home Integration

harikrishnan51688, 2021-03-26:
First thanks for your great work
I have successfully connected to Google home and Alexa. But the problem is when I say turn on the bulb the bulb "turn off". And vise versa. 
I use the code - switch - advanced  you provide. Help me out of this.

kakopappa, 2021-03-26:
GH has a different command format from Alexa. You have to hanlde it properly
https://github.com/kakopappa/sinric/blob/master/arduino_examples/google_home_switch_example.ino

You should consider moving to Sinric Pro https://sinric.pro. You just need to handle the SDK callbacks correctly once and it works.

harikrishnan51688, 2021-03-31:
I 

> GH has a different command format from Alexa. You have to hanlde it properly
> https://github.com/kakopappa/sinric/blob/master/arduino_examples/google_home_switch_example.ino
>
> You should consider [...]

Iam using sinric pro.  I use code in example switch-advanced but I press turn on in sinric dashboard the relay turn off. Also the same thing when I use alexa. Why?

sivar2311, 2021-03-31:
Depends on how your relay is connected (NO / NC).
You can invert the logic in sketch or change your wiring.
Do you have a circuit diagram you can share?

harikrishnan51688, 2021-03-31:
> Depends on how your relay is connected (NO / NC).
> You can invert the logic in sketch or change your wiring.
> Do you have a circuit diagram you can share?

[image]
I am using this circuit diagram. How I can invert the logic in sketch?

sivar2311, 2021-03-31:
Well... that's the diagram from example... and is not helpful here.
How are your devices connecte to the relays?? Do you use NO or NC wiring?
The example sketch is made for NO wiring!

harikrishnan51688, 2021-03-31:
> Well... that's the diagram from example... and is not helpful here.
> How are your devices connecte to the relays?? Do you use NO or NC wiring?
> The example sketch is made for NO wiring!

Iam using NC wiring. But I can use it perfectly before some days. What I can do?

sivar2311, 2021-03-31:
> Iam using NC wiring

Any reason for this? N/O is the usual wiring. Except you have a special reason for using N/C wiring.

> But I can use it perfectly before some days

This is strange, because we didn't change anything.

> What I can do?

1.) Change the wiring matching the sketch.
2.) Change the sketch matching the wiring.

For No. 2 you have to understand how the sketch is working. Do you?

harikrishnan51688, 2021-03-31:
> Any reason for this? N/O is the usual wiring. Except you have a special reason for using N/C wiring.
>
>
> This is strange, because we didn't change anything.
>
>
> 1.) Change the wiring matching the [...]

No. Can you give the code?

sivar2311, 2021-03-31:
Can you answer my questions?

sivar2311, 2021-03-31:
Ok, if you don't understand the code...change the wiring...

harikrishnan51688, 2021-03-31:
> Can you answer my questions?

Yes

harikrishnan51688, 2021-03-31:
> Ok, if you don't understand the code...change the wiring...

Iam using 4 channel relay board. Nodemcu D1 to relay D1 and so on. What can i change further?

harikrishnan51688, 2021-03-31:
> If you want to invert the relay output you have to invert the digitalWrite's

I don't know how the code works. Can you please give a sample code to invert the digitalWrite 's

sivar2311, 2021-03-31:
```digitalWrite(PIN, !state);```

harikrishnan51688, 2021-03-31:
I write 
digitalWrite(D1, HIGH); 
In void setup but it doesn't works

sivar2311, 2021-03-31:
To answer this question finally:
Change line 

https://github.com/sinricpro/esp8266-esp32-sdk/blob/d896bfffa554281c24399b804eadfb23eacb0a5e/examples/Switch/MultiSwitch_advance/MultiSwitch_advance.ino#L111

to 

```
 digitalWrite(relayPIN, !state);
```

harikrishnan51688, 2021-03-31:
Thanks for the great support. 
The problem was with my relay board. It is Inverting the signal HIGH TO LOW. After I connected D1 pin to led and it works well without any change in code. Thanks for your lovely reply!

harikrishnan51688, 2021-03-31:
> 
> 
> To answer this question finally:
> Change line
> 
> https://github.com/sinricpro/esp8266-esp32-sdk/blob/d896bfffa554281c24399b804eadfb23eacb0a5e/examples/Switch/MultiSwitch_advance/MultiSwitch_advance.ino#L111
> 
> to
> 
> ```
>  digitalWrite(relayPIN, !state);
> ```

Thanks for the great support.
Its works!!