kakopappa/sinric #480: not connecting to sinric server when add a piece of code

Perepeto, 2021-04-21:
I am testing my ESP8266 with the temperature sensor example provided with the library and it works fine but as soon as add some piece of code to read other kind of sensors then it does not connect anymore to sinric server.
It is the trace when it works, so when I add code the couple of lines at the end never appears.

15:38:12.806 -> [Wifi]: Connecting...................connected!
15:38:18.428 -> [WiFi]: IP-Address is 192.168.0.15
15:38:18.428 -> [SinricPro]: Device "6079fa72d9441c2b44abc45b" does not exist. Creating new device
15:38:18.521 -> [SinricPro:add()]: Adding device with id "6079fa72d9441c2b44abc45b".
15:38:18.615 -> [SinricPro:Websocket]: Connecting to WebSocket Server using SSL (ws.sinric.pro)
15:38:18.662 -> [SinricPro:Websocket]: headers: 
15:38:18.708 -> appkey:6af0a9fe-bf2d-4996-9a25-2ff40fdf29be
15:38:18.756 -> deviceids:6079fa72d9441c2b44abc45b
15:38:18.803 -> restoredevicestates:true
15:38:18.849 -> ip:192.168.0.15
15:38:18.849 -> mac:84:CC:A8:B4:16:00
15:38:18.849 -> platform:ESP8266
15:38:18.896 -> version:2.9.3
15:38:20.100 -> [SinricPro:Websocket]: connected
15:38:20.147 -> Connected to SinricPro

And below you can see my code, I only add  a function to read ultrasonic sensor and added a call in the loop, commenting the 3 last lines it works, but uncommenting only works reading the ultrasonic sensor.
Any idea whas is wrong ?

/*
 * Example for how to use SinricPro Temperaturesensor device:
 * - setup a temperature sensor device
 * - send temperature event to SinricPro server when temperature has changed
 * 
 * DHT Sensor is connected to D5 on ESP8266 devices / GPIO5 on ESP32 devices
 *
 * DHT Library used in this example: https://github.com/markruys/arduino-DHT
 * 
 * If you encounter any issues:
 * - check the readme.md at https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md
 * - ensure all dependent libraries are installed
 *   - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#arduinoide
 *   - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#dependencies
 * - open serial monitor and check whats happening
 * - check full user documentation at https://sinricpro.github.io/esp8266-esp32-sdk
 * - visit https://github.com/sinricpro/esp8266-esp32-sdk/issues and check for existing issues or open a new one
 */

// Uncomment the following line to enable serial debug output
#define ENABLE_DEBUG

#ifdef ENABLE_DEBUG
       #define DEBUG_ESP_PORT Serial
       #define NODEBUG_WEBSOCKETS
       #define NDEBUG
#endif 

#include <Arduino.h>
#ifdef ESP8266 
       #include <ESP8266WiFi.h>
#endif 
#ifdef ESP32   
       #include <WiFi.h>
#endif

#include "SinricPro.h"
#include "SinricProTemperaturesensor.h"
#include "DHT.h" // https://github.com/markruys/arduino-DHT

#define WIFI_SSID         "xxxxxxxxx"    
#define WIFI_PASS         "xxxxxxxxxxxxxxxxx"
#define APP_KEY           "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET        "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"
#define TEMP_SENSOR_ID    "xxxxxxxxxxxxxxxxxxxxxxxx"    // Should look like "5dc1564130xxxxxxxxxxxxxx"
#define BAUD_RATE         9600                // Change baudrate to your need (used for serial monitor)
#define EVENT_WAIT_TIME   60000               // send event every 60 seconds

#ifdef ESP8266
       #define DHT_PIN 5
#endif
#ifdef ESP32
       #define DHT_PIN 5
#endif

DHT dht;                                      // DHT sensor

bool deviceIsOn;                              // Temeprature sensor on/off state
float temperature;                            // actual temperature
float humidity;                               // actual humidity
float lastTemperature;                        // last known temperature (for compare)
float lastHumidity;                           // last known humidity (for compare)
unsigned long lastEvent = (-EVENT_WAIT_TIME); // last time event has been sent

//added for ultrasonic sensor
const int EchoPin = 12;
const int TriggerPin = 4;

/* bool onPowerState(String deviceId, bool &state) 
 *
 * Callback for setPowerState request
 * parameters
 *  String deviceId (r)
 *    contains deviceId (useful if this callback used by multiple devices)
 *  bool &state (r/w)
 *    contains the requested state (true:on / false:off)
 *    must return the new state
 * 
 * return
 *  true if request should be marked as handled correctly / false if not
 */
bool onPowerState(const String &deviceId, bool &state) {
  Serial.printf("Temperaturesensor turned %s (via SinricPro) \r\n", state?"on":"off");
  deviceIsOn = state; // turn on / off temperature sensor
  return true; // request handled properly
}

/* handleTemperatatureSensor()
 * - Checks if Temperaturesensor is turned on
 * - Checks if time since last event > EVENT_WAIT_TIME to prevent sending too much events
 * - Get actual temperature and humidity and check if these values are valid
 * - Compares actual temperature and humidity to last known temperature and humidity
 * - Send event to SinricPro Server if temperature or humidity changed
 */
void handleTemperaturesensor() {
  if (deviceIsOn == false) return; // device is off...do nothing

  unsigned long actualMillis = millis();
  if (actualMillis - lastEvent < EVENT_WAIT_TIME) return; //only check every EVENT_WAIT_TIME milliseconds

  temperature = random(1,30); //dht.getTemperature();          // get actual temperature in °C
//  temperature = dht.getTemperature() * 1.8f + 32;  // get actual temperature in °F
  humidity = random(1,100); //dht.getHumidity();                // get actual humidity

  if (isnan(temperature) || isnan(humidity)) { // reading failed... 
    Serial.printf("DHT reading failed!\r\n");  // print error message
    return;                                    // try again next time
  } 

  if (temperature == lastTemperature || humidity == lastHumidity) return; // if no values changed do nothing...

  SinricProTemperaturesensor &mySensor = SinricPro[TEMP_SENSOR_ID];  // get temperaturesensor device
  bool success = mySensor.sendTemperatureEvent(temperature, humidity); // send event
  if (success) {  // if event was sent successfuly, print temperature and humidity to serial
    Serial.printf("Temperature: %2.1f Celsius\tHumidity: %2.1f%%\r\n", temperature, humidity);
  } else {  // if sending event failed, print error message
    Serial.printf("Something went wrong...could not send Event to server!\r\n");
  }

  lastTemperature = temperature;  // save actual temperature for next compare
  lastHumidity = humidity;        // save actual humidity for next compare
  lastEvent = actualMillis;       // save actual time for next compare
}

// setup function for WiFi connection
void setupWiFi() {
  Serial.printf("\r\n[Wifi]: Connecting");
  WiFi.begin(WIFI_SSID, WIFI_PASS);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.printf(".");
    delay(250);
  }
  IPAddress localIP = WiFi.localIP();
  Serial.printf("connected!\r\n[WiFi]: IP-Address is %d.%d.%d.%d\r\n", localIP[0], localIP[1], localIP[2], localIP[3]);
}

// setup function for SinricPro
void setupSinricPro() {
  // add device to SinricPro
  SinricProTemperaturesensor &mySensor = SinricPro[TEMP_SENSOR_ID];
  mySensor.onPowerState(onPowerState);

  // setup SinricPro
  SinricPro.onConnected([](){ Serial.printf("Connected to SinricPro\r\n"); }); 
  SinricPro.onDisconnected([](){ Serial.printf("Disconnected from SinricPro\r\n"); });
  SinricPro.begin(APP_KEY, APP_SECRET);
  SinricPro.restoreDeviceStates(true); // get latest known deviceState from server (is device turned on?)
}

long ultrasound_read() {
pinMode(TriggerPin, OUTPUT);   
pinMode(EchoPin, INPUT);  
long duration, distanceCm;
   digitalWrite(TriggerPin, LOW);           //para generar un pulso limpio ponemos a LOW 4us
   delayMicroseconds(4);
   digitalWrite(TriggerPin, HIGH);          //generamos Trigger (disparo) de 10us
   delayMicroseconds(10);
   digitalWrite(TriggerPin, LOW);
   duration = pulseIn(EchoPin, HIGH);       //medimos el tiempo entre pulsos, en microsegundos
   distanceCm = duration * 10 / 292/ 2;   //convertimos a distancia, en cm
   return duration;
}

// main setup function
void setup() {
  Serial.begin(BAUD_RATE); Serial.printf("\r\n\r\n");
  dht.setup(DHT_PIN);

  setupWiFi();
  setupSinricPro();
}

void loop() {
  SinricPro.handle();
  handleTemperaturesensor();
/*  long x = ultrasound_read();
  Serial.print(" Ultrasound value = ");
  Serial.println(x);
*/
}

sivar2311, 2021-04-21:
Hi @Perepeto !
Unfortunately, you used the wrong repository for the issue. 
When it comes to SinricPro, please use this repository next time: https://github.com/sinricpro/esp8266-esp32-sdk

There are several issues in your code.
1. ```pinMode``` should be called only once, preferably in ```setup()```.
2. The main problem is that your function blocks the whole sketch. 
Moreover, your function is executed in each pass of the `loop()` function.

If you introduce a limit (e.g. read the value only once per second) everything works fine. 

See [Blink without delay](https://www.arduino.cc/en/Tutorial/BuiltInExamples/BlinkWithoutDelay)

Perepeto, 2021-04-21:
Hi Sivar2311, thanks for opening my eyes, after reading the link you shared I was able to run my code, not only the example, so also the global project with many kind of sensors, blynk and sinric all together.

About the repository, I am new here and undertood that this repository is for sinric topics but I was wrong. I am afraid that I continue without well understand the scope of this repository, if you can clarify will avoid to use a wrong repository in the future.

Thank you very much.

sivar2311, 2021-04-21:
>Hi Sivar2311, thanks for opening my eyes, after reading the link you shared I was able to run my code, not only the example, so also the global project with many kind of sensors, blynk and sinric all together.

👍 

>About the repository, I am new here and undertood that this repository is for sinric topics but I was wrong. I am afraid that I continue without well understand the scope of this repository, if you can clarify will avoid to use a wrong repository in the future.

Yeah it's a bit confusing because there is Sinric (sinric.com) and SinricPro (sinric.pro).
This is the library for SinricPro - check the comments at the beginning in each example:
```
If you encounter any issues:
...
visit https://github.com/sinricpro/esp8266-esp32-sdk/issues and check for existing issues or open a new one
```