kakopappa/sinric #483: SinricPro temperature/humidity sensors issue

pedrofnm, 2021-05-19:
Hello, lately I'm having some refresh problem with my temperature/humidity sensors. They only update every 4 hours or so. Sometimes it takes longer. Or if I reset the module. Dunno what's wrong with it. I used the code on the Arduino ide example (1min refresh time)and I'm using esp01 dht11 module. As you can see in the picture I reseted both sensors but only 1 is refreshing every minute and I bet that In a few minutes it will stop refreshing. Let me know if you need me to post the code here. 
[image]

sivar2311, 2021-05-20:
SinircPro is located here -> https://github.com/sinricpro

I assume both ESP's running the exact same sketch and have the same circumstances (good power supply, good WiFi signal strength). I also assume that these have already been checked by you. (If you have not checked this yet, please check these conditions first.)

Have you checked the activity-log? https://portal.sinric.pro/activitylog
Do you see any messages in the serial log?

pedrofnm, 2021-05-20:
Yes, both have good WiFi coverage, same code( except device ID), 1 amp power supply that is more than enough. The activity log only shows that It restore sensor state from the cloud when I reset the board, not updating by it self it seems
[image]

sivar2311, 2021-05-20:
Please enable serial log and check for any errors.