kakopappa/sinric #487: Webservice disconnected from sinric.com!

ModItBetter, 2021-08-01:
I tried to post this on your facebook but got deleted.
one unit that I had unplugged started giving me .. Webservice disconnected from sinric.com!

but my other 2 devices still worked fine.  I did a reset of my wifi and now all the units connected to sinric free is giving that error
is the server gone for free sinric? and odd the other devices worked up until I reset the wifi.. 
I also just tried on another wifi network and same error..

    I have been lazy and have not updated to sinric pro or my youtube video files as .
I've only had about 3k total views but probably should update the steps and files for pro .

kakopappa, 2021-08-01:
Have you tried connecting via mobile hotspot?

ModItBetter, 2021-08-03:
Hi thanks for quick reply .. this is only effecting my nodes esp8266 running sinric on that one router..
I'm now 100 percent sure this has nothing to do with you.    I logged into one of my wifi camera's and did the built in wifi scan and it found 2 ssids with the same name as my router .. One is mine(alexa devices only router) the other  I have no clue it even has the same password but it does not have any internet ..
So either a neighbor is doing something sketchy or one of my very many devices has some how turned it self into a hotspot with the info I use to log into my alexa device only router..

If anybody else runs into issue's with your nodemcu stuck on the wrong router / clone or what ever .. 
put this under void setup...  if it refuses to pick up the new ip 
WiFi.disconnect(true);

anyways thanks for taking time with suggestion and sorry for killing your time with something that is on my end..