kakopappa/sinric #490: sinric dashboard is saying device is offline even if my project is connected to internet

sushant96702, 2021-09-09:
sinric dashboard is saying device is offline even if my project is connected to internet and because it is offline i am not able to use my devices

/**********************************************************************************
    TITLE: Google + Alexa + Manual Switch/Button control 4 Relays using NodeMCU & Sinric Pro (Real time feedback)
    (flipSwitch can be a tactile button or a toggle switch) (code taken from Sinric Pro examples then modified)
    Click on the following links to learn more.
    YouTube Video: https://youtu.be/gpB4600keWA
    Related Blog : https://iotcircuithub.com/esp8266-projects/
    by Tech StudyCell
    Preferences--> Aditional boards Manager URLs :
    https://dl.espressif.com/dl/package_esp32_index.json, http://arduino.esp8266.com/stable/package_esp8266com_index.json

    Download Board ESP8266 NodeMCU : https://github.com/esp8266/Arduino
    Download the libraries
    ArduinoJson Library: https://github.com/bblanchon/ArduinoJson
    arduinoWebSockets Library: https://github.com/Links2004/arduinoWebSockets
    SinricPro Library: https://sinricpro.github.io/esp8266-esp32-sdk/

    If you encounter any issues:
   - check the readme.md at https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md
   - ensure all dependent libraries are installed
     - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#arduinoide
     - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#dependencies
   - open serial monitor and check whats happening
   - check full user documentation at https://sinricpro.github.io/esp8266-esp32-sdk
   - visit https://github.com/sinricpro/esp8266-esp32-sdk/issues and check for existing issues or open a new one
 **********************************************************************************/

// Uncomment the following line to enable serial debug output
//#define ENABLE_DEBUG

#ifdef ENABLE_DEBUG
#define DEBUG_ESP_PORT Serial
#define NODEBUG_WEBSOCKETS
#define NDEBUG
#endif

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "SinricPro.h"
#include "SinricProSwitch.h"

#include <map>

#define WIFI_SSID         "JioFiber799_2"
#define WIFI_PASS         "11223344"
#define APP_KEY           "c167fce3-53ac-4f04-8f2dxxxxxxxxx-ab3"      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET        "9ba0d59f-e0c9-4f64-9xxxxxxxxxxxxxxxxxxxxxxxxxxx-0236-435e-aa46-12a301266c56"   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"

//Enter the device IDs here
#define device_ID_1   "60e92aaxxxxxxxxxxxxx62c47c4d"
#define device_ID_2   "60e92a88xxxxxxxxxxxx2c47c4b"
#define device_ID_3   "xxxxxxxxxxxxxxxxxxxxxxxx"
#define device_ID_4   "60e92axxxxxxxxxxxxxxd5b0cf7be"

// define the GPIO connected with Relays and switches
#define RelayPin1 5  //D1
#define RelayPin2 4  //D2
#define RelayPin3 14 //D5
#define RelayPin4 12 //D6

#define SwitchPin1 10  //SD3
#define SwitchPin2 0   //D3 
#define SwitchPin3 13  //D7
#define SwitchPin4 3   //RX

#define wifiLed   16   //D0

// comment the following line if you use a toggle switches instead of tactile buttons
#define TACTILE_BUTTON 1

#define BAUD_RATE   9600

#define DEBOUNCE_TIME 250

typedef struct {      // struct for the std::map below
  int relayPIN;
  int flipSwitchPIN;
} deviceConfig_t;

// this is the main configuration
// please put in your deviceId, the PIN for Relay and PIN for flipSwitch
// this can be up to N devices...depending on how much pin's available on your device ;)
// right now we have 4 devicesIds going to 4 relays and 4 flip switches to switch the relay manually
std::map<String, deviceConfig_t> devices = {
  //{deviceId, {relayPIN,  flipSwitchPIN}}
  {device_ID_1, {  RelayPin1, SwitchPin1 }},
  {device_ID_2, {  RelayPin2, SwitchPin2 }},
  {device_ID_3, {  RelayPin3, SwitchPin3 }},
  {device_ID_4, {  RelayPin4, SwitchPin4 }}
};

typedef struct {      // struct for the std::map below
  String deviceId;
  bool lastFlipSwitchState;
  unsigned long lastFlipSwitchChange;
} flipSwitchConfig_t;

std::map<int, flipSwitchConfig_t> flipSwitches;    // this map is used to map flipSwitch PINs to deviceId and handling debounce and last flipSwitch state checks
// it will be setup in "setupFlipSwitches" function, using informations from devices map

void setupRelays() {
  for (auto &device : devices) {           // for each device (relay, flipSwitch combination)
    int relayPIN = device.second.relayPIN; // get the relay pin
    pinMode(relayPIN, OUTPUT);             // set relay pin to OUTPUT
    digitalWrite(relayPIN, HIGH);
  }
}

void setupFlipSwitches() {
  for (auto &device : devices)  {                     // for each device (relay / flipSwitch combination)
    flipSwitchConfig_t flipSwitchConfig;              // create a new flipSwitch configuration

    flipSwitchConfig.deviceId = device.first;         // set the deviceId
    flipSwitchConfig.lastFlipSwitchChange = 0;        // set debounce time
    flipSwitchConfig.lastFlipSwitchState = true;     // set lastFlipSwitchState to false (LOW)--

    int flipSwitchPIN = device.second.flipSwitchPIN;  // get the flipSwitchPIN

    flipSwitches[flipSwitchPIN] = flipSwitchConfig;   // save the flipSwitch config to flipSwitches map
    pinMode(flipSwitchPIN, INPUT_PULLUP);                   // set the flipSwitch pin to INPUT
  }
}

bool onPowerState(String deviceId, bool &state)
{
  Serial.printf("%s: %s\r\n", deviceId.c_str(), state ? "on" : "off");
  int relayPIN = devices[deviceId].relayPIN; // get the relay pin for corresponding device
  digitalWrite(relayPIN, !state);             // set the new relay state
  return true;
}

void handleFlipSwitches() {
  unsigned long actualMillis = millis();                                          // get actual millis
  for (auto &flipSwitch : flipSwitches) {                                         // for each flipSwitch in flipSwitches map
    unsigned long lastFlipSwitchChange = flipSwitch.second.lastFlipSwitchChange;  // get the timestamp when flipSwitch was pressed last time (used to debounce / limit events)

    if (actualMillis - lastFlipSwitchChange > DEBOUNCE_TIME) {                    // if time is > debounce time...

      int flipSwitchPIN = flipSwitch.first;                                       // get the flipSwitch pin from configuration
      bool lastFlipSwitchState = flipSwitch.second.lastFlipSwitchState;           // get the lastFlipSwitchState
      bool flipSwitchState = digitalRead(flipSwitchPIN);                          // read the current flipSwitch state
      if (flipSwitchState != lastFlipSwitchState) {                               // if the flipSwitchState has changed...
#ifdef TACTILE_BUTTON
        if (flipSwitchState) {                                                    // if the tactile button is pressed
#endif
          flipSwitch.second.lastFlipSwitchChange = actualMillis;                  // update lastFlipSwitchChange time
          String deviceId = flipSwitch.second.deviceId;                           // get the deviceId from config
          int relayPIN = devices[deviceId].relayPIN;                              // get the relayPIN from config
          bool newRelayState = !digitalRead(relayPIN);                            // set the new relay State
          digitalWrite(relayPIN, newRelayState);                                  // set the trelay to the new state

          SinricProSwitch &mySwitch = SinricPro[deviceId];                        // get Switch device from SinricPro
          mySwitch.sendPowerStateEvent(!newRelayState);                            // send the event
#ifdef TACTILE_BUTTON
        }
#endif
        flipSwitch.second.lastFlipSwitchState = flipSwitchState;                  // update lastFlipSwitchState
      }
    }
  }
}

void setupWiFi()
{
  Serial.printf("\r\n[Wifi]: Connecting");
  WiFi.begin(WIFI_SSID, WIFI_PASS);

  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.printf(".");
    delay(250);
  }
  digitalWrite(wifiLed, LOW);
  Serial.printf("connected!\r\n[WiFi]: IP-Address is %s\r\n", WiFi.localIP().toString().c_str());
}

void setupSinricPro()
{
  for (auto &device : devices)
  {
    const char *deviceId = device.first.c_str();
    SinricProSwitch &mySwitch = SinricPro[deviceId];
    mySwitch.onPowerState(onPowerState);
  }

  SinricPro.begin(APP_KEY, APP_SECRET);
  SinricPro.restoreDeviceStates(true);
}

void setup()
{
  Serial.begin(BAUD_RATE);

  pinMode(wifiLed, OUTPUT);
  digitalWrite(wifiLed, HIGH);

  setupRelays();
  setupFlipSwitches();
  setupWiFi();
  setupSinricPro();
}

void loop()
{
  SinricPro.handle();
  handleFlipSwitches();
}

kakopappa, 2021-09-09:
1. Are you using Sinric.com or Sinric.Pro?

2. Have you tried connecting to server via Mobile hostspot and does it
still show offline ?

3. Can you share a image of the devices are online after you enable logs
(Arduino Serial)

sushant96702, 2021-09-17:
1)Sinric Pro
2) sometimes it work and sometimes it doesnt like the same prob like above ( this is a home project so it is not possible to upload code again and again
3)this time it is working when it doesnt i will send you pic
   Workin time `https://pastebin.pl/view/9ee26aa4`

sushant96702, 2021-09-19:
> 1. Are you using Sinric.com or Sinric.Pro? 2. Have you tried connecting to server via Mobile hostspot and does it still show offline ? 3. Can you share a image of the devices are online after you enable logs (Arduino Serial)
> […](#)

whenever i upload same code again it works flawlessly

kakopappa, 2021-09-19:
I looked at the server log for 60e92aae5c24133d62c47c4d.. it seems your devices get disconnected after some time and does not connect again.

1. Try setting up connect/disconnect push notifications so you would know when the device gets disconnected
2. https://microcontrollerslab.com/reconnect-esp32-to-wifi-after-lost-connection/

sivar2311, 2021-09-19:
You can also try to use `setAutoReconnect(true)` after you connected successfully to your wifi network:

```C++
void setupWiFi() {
    // connect to wifi
    WiFi.begin(SSID, PASS);
    while (WiFi.status() =! WL_CONNECTED) {
        Serial.print(".");
        delay(250);
    }
    // now wifi is connected...
    Serial.println("connected!");
    // setAutoReconnect(true) will reconnect wifi when connection is lost
    // (must be called after wifi connection was established!)
    WiFi.setAutoReconnect(true);
}
```

sushant96702, 2021-09-21:
> I looked at the server log for 60e92aae5c24133d62c47c4d.. it seems your devices get disconnected after some time and does not connect again.
> 
> 1. Try setting up connect/disconnect push notifications so you would know when the device gets disconnected
> 2. https://microcontrollerslab.com/reconnect-esp32-to-wifi-after-lost-connection/
> 
> BTW: can't read your code above. try adding code formatting

this is what serial monitor prints when my prject doesnt work

13:57:18.072 -> ........connected!
13:57:20.804 -> [WiFi]: IP-Address is 192.168.1.100
13:57:20.851 -> [SinricPro]: Device "60e92a885c24133d62c47c4b" does not exist. Creating new device
13:57:20.945 -> [SinricPro:add()]: Adding device with id "60e92a885c24133d62c47c4b".
13:57:21.040 -> [SinricPro]: Device "60e92aae5c24133d62c47c4d" does not exist. Creating new device
13:57:21.134 -> [SinricPro:add()]: Adding device with id "60e92aae5c24133d62c47c4d".
13:57:21.180 -> [SinricPro]: Device "60e92acfaf7d763d5b0cf7be" does not exist. Creating new device
13:57:21.275 -> [SinricPro:add()]: Adding device with id "60e92acfaf7d763d5b0cf7be".
13:57:21.368 -> [SinricPro]: Device "xxxxxxxxxxxxxxxxxxxxxxxx" does not exist. Creating new device
13:57:21.416 -> [SinricPro:add()]: Adding device with id "xxxxxxxxxxxxxxxxxxxxxxxx".
13:57:21.510 -> [SinricPro:Websocket]: Connecting to WebSocket Server using SSL (ws.sinric.pro)
13:57:21.604 -> [SinricPro:Websocket]: headers: 
13:57:21.604 -> appkey:c167fce3-53ac-4f04-8f2d-ab34ba1603cd
13:57:21.651 -> deviceids:60e92a885c24133d62c47c4b;60e92aae5c24133d62c47c4d;60e92acfaf7d763d5b0cf7be;xxxxxxxxxxxxxxxxxxxxxxxx
13:57:21.792 -> restoredevicestates:true
13:57:21.840 -> ip:192.168.1.100
13:57:21.840 -> mac:50:02:91:E1:AE:E5
13:57:21.840 -> platform:ESP8266
13:57:21.886 -> version:2.9.9

sushant96702, 2021-09-21:
```
#define ENABLE_DEBUG

#ifdef ENABLE_DEBUG
#define DEBUG_ESP_PORT Serial
#define NODEBUG_WEBSOCKETS
#define NDEBUG
#endif

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "SinricPro.h"
#include "SinricProSwitch.h"

#include <map>

#define WIFI_SSID         "JioFiber799_2"
#define WIFI_PASS         "11223344"
#define APP_KEY           "c167fce3-53ac-4f04-8f2d-ab34ba1603cd"      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET        "9ba0d59f-e0c9-4f64-9815-d1618bcf78ba-121c4792-0236-435e-aa46-12a301266c56"   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"

//Enter the device IDs here
#define device_ID_1   "60e92aae5c24133d62c47c4d"
#define device_ID_2   "60e92a885c24133d62c47c4b"
#define device_ID_3   "xxxxxxxxxxxxxxxxxxxxxxxx"
#define device_ID_4   "60e92acfaf7d763d5b0cf7be"

// define the GPIO connected with Relays and switches
#define RelayPin1 5  //D1
#define RelayPin2 4  //D2
#define RelayPin3 14 //D5
#define RelayPin4 12 //D6

#define SwitchPin1 10  //SD3
#define SwitchPin2 0   //D3 
#define SwitchPin3 13  //D7
#define SwitchPin4 3   //RX

#define wifiLed   16   //D0

// comment the following line if you use a toggle switches instead of tactile buttons
#define TACTILE_BUTTON 1

#define BAUD_RATE   9600

#define DEBOUNCE_TIME 250

typedef struct {      // struct for the std::map below
  int relayPIN;
  int flipSwitchPIN;
} deviceConfig_t;

// this is the main configuration
// please put in your deviceId, the PIN for Relay and PIN for flipSwitch
// this can be up to N devices...depending on how much pin's available on your device ;)
// right now we have 4 devicesIds going to 4 relays and 4 flip switches to switch the relay manually
std::map<String, deviceConfig_t> devices = {
  //{deviceId, {relayPIN,  flipSwitchPIN}}
  {device_ID_1, {  RelayPin1, SwitchPin1 }},
  {device_ID_2, {  RelayPin2, SwitchPin2 }},
  {device_ID_3, {  RelayPin3, SwitchPin3 }},
  {device_ID_4, {  RelayPin4, SwitchPin4 }}
};

typedef struct {      // struct for the std::map below
  String deviceId;
  bool lastFlipSwitchState;
  unsigned long lastFlipSwitchChange;
} flipSwitchConfig_t;

[... 99 lines cut ...]
  }

  SinricPro.begin(APP_KEY, APP_SECRET);
  SinricPro.restoreDeviceStates(true);
}

void setup()
{
  Serial.begin(BAUD_RATE);

  pinMode(wifiLed, OUTPUT);
  digitalWrite(wifiLed, HIGH);

  setupRelays();
  setupFlipSwitches();
  setupWiFi();
  setupSinricPro();
}

void loop()
{
  SinricPro.handle();
  handleFlipSwitches();
}
```

sushant96702, 2021-09-21:
> ```c++
>    Serial.println("connected!");
>     // setAutoReconnect(true) will reconnect wifi when connection is lost
>     // (must be called after wifi connection was established!)
>     WiFi.setAutoReconnect(true);
> ```

How do i keep my wifi led on as esp8266 gets connected to internet

sivar2311, 2021-09-21:
I will look at your code later (when I have more time) in detail.
But what I notice immediately is the Invalid DeviceID "xxxx....".
Please delete it or insert a valid DeviceID!